COVESA/dlt-viewer #222: Support for opening / drag and drop of (multiple) ZIP DLT files

endzone, 2021-09-13:
In our project DLT traces files are being stored in compressed standard ZIP file format , due to high compression ratio that can be achieved on regular DLT files. Traces are being stored on network drives with read-only access to standard developer.

Currently / AS IS:  In case a sequence of multiple ZIP DLTs shall be loaded into DLT viewer, the files need to be copied from network read-only drive to local disc, then unzipped and then DLT files can be drag+dropped into DLT viewer.

Feature Request / TO BE:  I would like to ask DLT Viewer team to support standard ZIP format for ZIP compressed DLT files in "Open  File" dialog as well as Drag-and-Drop Feature. 

This way, in our project it would be possible to open multi-file Zipped DLT traces right from read-only network storage. The above manual steps of "copy ZIP + unzip ZIP to DLT + load DLT to DLT viewer" could be replaced by just a single step of "load multiple ZIP DLTs into DLT viewer" and save a lot of time.

Other trace analysis tools for similar tasks do support loading multiple zip files already.

alexmucde, 2021-11-23:
Opening DLT files must be done on a local storage. This is because the DLT Viewer does not load a DLT file into memory, it reads the DLT messages directly from the storage.
But it should be possible that the users opens a zip file on a network storage and the DLT files are unzipped to a temporary directory on the local storage.

Qt does not provide any feature to unzip a file.

Possible solutions found in the internet:
https://github.com/stachenov/quazip
https://zlib.net/

If someone has a solution, please provide some hints or a PR.

alexmucde, 2022-05-07:
One idea would be to have a local temporary file and when drag and drop the DLT Viewer calls an external zip tool to unzip the files into the temporary folder and then load the unzipped files.
Please look also at the newly integrated file explorer feature. This would be an alternative for drag an drop.

Any contributer to provide a PR for this feature request?

Usha-Rajanna, 2022-10-24:
@endzone as Alex mentioned Qt does not provide any inbuilt function to uncompress the .zip file.
Only way is to use third party .dll files called quazip and zlib. But due to the restrictions of using third party dll's
We have stopped this enhancement for now.

If any one can suggest any other Qt inbuilt way to unzip the file. We would surely take this up in future enhancement.

alexmucde, 2022-12-12:
Qt does not provide any class/library to unzip files. Will not integrate any dependency to external libraries. DLT files must be unziped before opened with DLT Viewer.

closed 2022-12-12 by alexmucde