COVESA/dlt-viewer #232: win: recorded traces have non monotonic timestamps

mbehr1, 2021-11-26:
DLT traces recorded with dlt-viewer under windows have non monotonic - going backwards - timestamps.

See here the problematic code snippet:

https://github.com/COVESA/dlt-viewer/blob/eaf1571483c59c0448fcd4bf99da5f6b7c075e9a/src/mainwindow.cpp#L3585

and a small test app:
```
#include <cstdint>
#include <windows.h>
#include <time.h>
 
#include <iostream>
 
struct TestTime {
    time_t seconds;
    int32_t microseconds;
};
 
int main(){
 
    TestTime str;
    bool isMonotonic = true;
    uint64_t lastT = 0;
    do{
        // code from dltviewer
        SYSTEMTIME systemtime;
        GetSystemTime(&systemtime);
        time_t timestamp_sec;
        time(&timestamp_sec);
        str.seconds = (time_t)timestamp_sec;
        str.microseconds = (int32_t)systemtime.wMilliseconds * 1000; // for some reasons we do not have microseconds in Windows !
        // code from dltviewer end
 
        uint64_t timeInMicros = str.microseconds + (str.seconds * 1000000ul);
        isMonotonic = timeInMicros >= lastT;
        if (!isMonotonic) {
            std::cout << "got " << str.seconds << "." << str.microseconds << " " << timeInMicros << std::endl;
            std::cout << "lastT was " << lastT << " dist=" << (lastT-timeInMicros) << std::endl;
        }else {
            lastT = timeInMicros;
        }
    }while(isMonotonic);
}
``` 

example output is:
```
got 1637922286.0 1637922286000000
lastT was 1637922286992000 dist=992000
```

mbehr1, 2021-11-26:
a fix should/could be to use timespec_get(...) which seems to be available under win (since VS 2015) as well and provides tv_sec and tv_nsec.

mbehr1, 2021-11-26:
@alexmucde @gernotmuc  fyi.

alexmucde, 2023-09-29:
@mbehr1 Do you have a fix/PR for this?

mbehr1, 2023-09-29:
Hi Alex,

I can prepare one. (I pasted the code snippet already in the comments.)

(will take a few days)

mbehr1, 2023-10-03:
@alexmucde see PR#381

closed 2023-10-05 by alexmucde