COVESA/dlt-viewer #234: Copy selected messages as Jira table?

SvenLuebke, 2021-12-16:
Dear dlt-viewer developers!

Is a clipboard export of selected Dlt messages formatted as Jira table (in context menu) too specialized? I could provide a PR for that if somebody is interested...

BR,
Sven

alexmucde, 2021-12-17:
Hi Sven,

yes this feature looks interesting, this is a common use case. please provide a PR.

    Alex

alexmucde, 2022-05-07:
@SvenLuebke Is there still an activity form your side? If not i will close this issue.

SvenLuebke, 2022-05-07:
@alexmucde Please don't close this issue! I have to discuss this with the company I'm working at and didn't have time to do it yet.

SvenLuebke, 2022-06-09:
@alexmucde Are you able to explain the following text?:

Every contribution must carry the following sign-off line with your real name and email address:
"Signed-off-by: Firstname Lastname <[firstname.lastname@example.com](mailto:firstname.lastname@example.com)>"

Where do I have to add this sign-off line?

SvenLuebke, 2022-06-15:
Created [PR#276](https://github.com/COVESA/dlt-viewer/pull/276)

closed 2022-06-15 by SvenLuebke