COVESA/dlt-viewer #242: bool args are persisted without DLT_TYLE_8BIT

mbehr1, 2022-01-22:
Bool args are persisted with a TYLE of 0 which is not fulfilling
dlt autosar spec: [Dlt139] ⌈Type Length (TYLE) shall be 1. ⌋

This causes e.g. the Wireshark DLT dissector to fail on parsing the payload.

Wrong code is here:
https://github.com/COVESA/dlt-viewer/blob/03baa67d3bb059458cb5b8e0ed940ea6f607f575/qdlt/qdltargument.cpp#L312

I guess a | DLT_TYLE_8BIT would be sufficient to fix.

Shall I send a PR?

mbehr1, 2022-01-22:
@alexmucde @gernotmuc fyi.

alexmucde, 2022-01-24:
@mbehr1 

Yes that's right. I would change the following line 333:

if((typeInfo == DltTypeInfoSInt) || (typeInfo == DltTypeInfoUInt) || (typeInfo == DltTypeInfoFloa) **|| (typeInfo == DltTypeInfoBool)**) {

If data size is 1, this would add dltType |= DLT_TYLE_8BIT;

Yes you can provide a PR.

mbehr1, 2022-01-26:
fixed wit #247

closed 2022-01-26 by mbehr1