COVESA/dlt-viewer #243: Only one line can be copied when multiple lines are selected

xRowe, 2022-01-24:
Dear Developers

I got an issue while using the latest release 2.13.3. Windows build
Could you please check if it is an issue or not?

[Steps]
1. Select multiple lines message in the main window.
2. Right Click, and select "Copy Selection to Clipboard" or "Copy Selection Payload to Clipboard"
3. Try to paste the content in Notepad++

[Result]
Only the first selection line can be copied and pasted.

ps. While doing these in the search windows, all the selections can be copied and pasted correctly.

alexmucde, 2022-01-26:
@xRowe I cannot reproduce this issue in Windows. Perhaps it depends on OS or Qt version.

There was a similar issue #224 , where a wrong dll was used.

Please check again.

xRowe, 2022-02-01:
Hello @alexmucde 

Sorry, late reply, I do not usually focus on github. 
Thanks for your quick reply.

Yes, I searched issue and found #224, ever I suspect it is a similar issue.
#224 is an issue that could not copy multiple lines in the **Search Window** via  (Ctrl-C or context menu "Copy Selection (Payload) to Clipboard"), but mine is working correctly.

Meanwhile, in the **Main Window**, select multiple lines, and copy via (Ctrl-C), it is working perfectly. But via context menu "Copy Selection (Payload) to Clipboard"), it does not -> (Only the first (lowest Index) of the selected lines is copied).
Even if using the one I compiled the dlt-viewer via github-ACTION or in my laptop, I got the same result.

So, I do not know how to debug it.

Thanks,
Rowe

alexmucde, 2022-02-01:
It looks like it has to do with a specific combination of Qt version and MSVC version.
With Qt 5.12.12 and MSVC2017 i got this issue.
WIth Qt 5.12.12 and MSVC2015 i do not have this issue.
With Qt 5.15.2 and MSVC2019 i do not have this issue.

alexmucde, 2022-02-01:
With Qt 5.12.12 and MSVC2017 i do not get this issue in Debug build.

alexmucde, 2022-02-01:
Found uninitialised variables in the export function. I will provide a PR.

alexmucde, 2022-02-01:
Fixed by #251

closed 2022-02-02 by alexmucde