COVESA/dlt-viewer #261: Release artifacts not present in releases page, is this expected?

GianlucaAgl, 2022-03-10:
It seems from changelogs and closed PRs that compiled binaries should be present in GitHub release page, but currently no binary is listed there https://github.com/COVESA/dlt-viewer/releases

Eg.
[image]

I was expecting to find the binary after having seen PR #187, that was integrated in Release v2.21.2 . Also the install.md file in scripts/ folder reports that, for example in the windows folder (at least in master branch):
https://github.com/COVESA/dlt-viewer/blob/master/scripts/windows/install.md

Is this currently only a plan for the next release? Or is this an error?

I have not seen any official announcement, so this is just a question to understand if it's possible to obtain the binaries without dealing with Qt framework SDK, now or in the future.

Thank you,
Gianluca

alexmucde, 2022-04-02:
@GianlucaAgl We use Github Action to test PR builds. Release Artifacts should be also generated, but this is currently not working. Need some help to fix this, as i am not familiar with this.

alexmucde, 2022-10-14:
Any help from the community?

Usha-Rajanna, 2022-10-28:
@GianlucaAgl  Sorry we have not planned to release any binaries for now. But we have CI artifacts available. Please check the attachements. Thank you.
[image]
[image]
[image]

Usha-Rajanna, 2022-10-28:
Wont be fixed

closed 2022-10-28 by Usha-Rajanna