COVESA/dlt-viewer #262: Ability for send serial message in Serial ASCII Mode?

beidouchangxi, 2022-04-01:
With Serial ASCII Mode I can actually use DLT viewer as a powerful serial debug tool.
The only downside is I cannot send a message by DLT viewer.
Is it possible for add this feature?

beidouchangxi, 2022-04-02:
Yes, I mean provide a dialog for send commands to ECU in Serial ASCII Mode, a plugin will be pretty enough to do it.
But I took a look on the API documentm, it seems like no API can provide this ability?

Another idea is use the existing "Send Injection" dialog, which is currently doing nothing in Serial ASCII Mode.

beidouchangxi, 2022-04-02:
I trying to look into code about how to do it but I am not familar with cpp and QT,
If I am lucky maybe I can provide a PR, are you interested in this PR?

beidouchangxi, 2022-04-02:
a PR is created based on "Send Injection" dialog

alexmucde, 2022-04-02:
Great thanks

alexmucde, 2022-05-07:
PR is integrated

closed 2022-05-07 by alexmucde