COVESA/dlt-viewer #280: Filter color setting is not applied when the second color is same APP (application) ID, but different CT (context) ID

firedevil79, 2022-07-19:
[image]

precondition:
there are multiple filters in the project

Filter 1:  (all Tuner messages shall be in red)
color setting: red
AppID: TUNE
CT ID: empty 

Filter 2: (Tuner API shall be shown in green)
color setting: green
AppID: TUNE
CT ID; API

check DLT

only Filter 1 color is shown

I would expect the Filter 2 can also be added so I can see all messages for TUNE on green, but TUNE -> API in red

note: 
when the setting is as following:
Filter 1:  (all Tuner MAIN messages shall be in red)
color setting: red
AppID: TUNE
CT ID: MAIN 

Filter 2: (Tuner API shall be shown in green)
color setting: green
AppID: TUNE
CT ID; API

then the color setting is as expected, but I cannot see the "other" Messages but API and MAIN

eengnr, 2022-07-28:
Filters are applied from bottom to top. That's why it works if your Filter 2 is above Filter 1.

alexmucde, 2022-10-14:
@firedevil79 Marking of messages depends on the order. Thefirst filter which matches from top wins.

closed 2022-12-12 by alexmucde