COVESA/dlt-viewer #285: Make is not working in the new release

Halfgeek0, 2022-08-11:
The make command gives the below error in the latest release
commit SHA 9206b0b4ec41a134de02cbab28d35c2da24f241c

../../src/mainwindow.cpp: In lambda function:
../../src/mainwindow.cpp:6748:47: error: no matching function for call to ‘QProcess::startDetached(qint64*)’
                     process.startDetached(&pid);

alexmucde, 2022-10-14:
@Halfgeek0 We do not know any issues. Is it possible that you are using Qt6, which is not supported yet?

Usha-Rajanna, 2022-10-21:
@Halfgeek0 can you please specify the platform you are using,  if Linux please use 20.04 Ubuntu

Usha-Rajanna, 2022-10-21:
@Halfgeek0 Can you please tell us the platform you are using , if it is Linux use 20.04 Ubuntu.
I hope this should solve your problem.

alexmucde, 2022-12-12:
No answer, consider it is already solved. Please reopen if needed.

closed 2022-12-12 by alexmucde

tt1m0n, 2022-12-15:
I have same problem. How possible fix this issues using Ubuntu 18.04

Usha-Rajanna, 2022-12-15:
@tt1m0n As we have updated the Source to Support 20.04 and very soon it will be supporting  QT 6, please update both qt and Linux version. 
Thank you