COVESA/dlt-viewer #291: [MacOS] Plugins makes DLT crash on start

carlossilva2, 2022-10-13:
When building from source on Mac, if there are plugins on the package the App doesn't start and keeps crashing. 

Did I miss anything? 
Is any step missing on the Readme/Install file?

alexmucde, 2022-10-14:
@carlossilva2 Could it be that you have somewhere old plugins which are not matching the Qt version or the DLT Viewer version. This could cause crashes.

carlossilva2, 2022-10-16:
I don't think so.

I'm using qt5, cmake and qt-creator to generate the image file.
`mkdir build && cd build && Qt5_DIR="/usr/local/opt/qt5" cmake -S .. -B . && sudo make install`.
Anything missing from here?

alexmucde, 2022-10-21:
The build should work, but there could be some old plugins somewhere in the filesystems, which are then tried to be loaded with wrong or old build system. Please try to cleanup whole filesystem first before building.

closed 2022-10-26 by alexmucde