COVESA/dlt-viewer #294: Try to connect ECU on serial port "\\.\COM38"

khs7196, 2022-10-26:
Hi Developer!

I'm trying to connect ECU thru serial port on Window but I have no idea how to use it.

**Port:** \\.\COM38
**Baudrate:** 115200

Could you please let me know how to use serial port thing?

Thank you

alexmucde, 2022-10-27:
@khs7196 Yes this should work, you can also simply type COM38. Lattest version on Github will provide an improved possibility for selection.

khs7196, 2022-10-27:
Thank you for feedback!!

I think I'm already trying with latest version but it is not working.

If the debugging board has id/pw, should I type it differently? For example, COM38\root\root?

alexmucde, 2022-10-27:
I used serial connections all time and it was working fine. COM38 should be sufficient. Must be another issue with your PC.

alexmucde, 2022-12-12:
Please reopen, if there is still an issue.

closed 2022-12-12 by alexmucde