COVESA/dlt-viewer #297: Compiled version (windows 32/64bit and Linux)

bovirus, 2022-11-06:
@Usha-Rajanna 

Could you please consider to provide in "Release" tag not only source code but also compiled vesrion (for Windows 32 and 64bit and Linxu)?

It's an important feature for people that don't have knowledge/resources to compile your source code.

Thanks.

Usha-Rajanna, 2022-11-06:
@bovirus Sorry at the moment we cannot agree for the Release tags, but I will help you to resolve any issue You face while building the dlt-viewer. 
Please let me know if you are facing some issue to compile the Source code.

bovirus, 2022-11-06:
@Usha-Rajanna

I compield version 22.0. and 213.0 (beta) in my configuratyion.

Some note:

Your source is based on Qt 5.15.2 and seems didn't work with Qt 6.4.0 (latest offciail build).

Then I insttaled in my system QT 5.15.2.

You build_connfig.bat is based on Visual C+ not free ("Enterprise) an not on other version (ex. free VC++ "Community" version).

To work with my configuration (Visual C+ Community 2022 64bit) I should set some variables in config_build.bat

set QWT=6.1.4
SET QTVER=5.15.2
SET QTDIR=C:\Qt\%QTVER%\msvc2019_64

SET MSVC_VER=2022
SET MSVC_DIR=C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build

SET PATH=%QTDIR%\bin;%MSVC_DIR%;%PATH%

SET DLT_VIEWER_SDK_DIR=%CD%\_DltViewerSDK
 
I also setup the DL_VIEWER_SDK variable as sub-fodler of current source folder.

Usha-Rajanna, 2022-11-07:
@bovirus use QT 5.12.12 and  VS 2017 community is free and while installing add windows sdk 10 and include c++/c package
QWT is not needed to build the source code.
QTVER=512.12
QTDIR=C:\QT\%QTVER%\msvc%MSVC_VER%_64
Check the path of QT and VS path is correct in your case.
SDK path give complete path if you have not set the current directory.

and DLt will not support QT 6 and above version.

alexmucde, 2022-12-12:
Duplicate of #277

closed 2022-12-12 by alexmucde