COVESA/dlt-viewer #299: Linefeeds or Cariage Return in corrupted DLT causes CSV export inconsistency

tomasdrozdskodaauto, 2022-11-21:
Some corrupted DLT files might contain some Linefeeds or Cariage Return for example in EcuId, AppId or CtxId. This causes a line wrap. When exporting to CSV, the number of columns is not correct in some lines. This makes the CSV unusable in some applications.

Perhaps during Export all Linefeeds and Carriage Return should be removed.

[image]

[image]

alexmucde, 2022-12-12:
Must be fixed by removing linefeeds and cariage returns in EcuId, AppId and CtxId before Export.

alexmucde, 2023-09-29:
@tomasdrozdskodaauto Found solution please check attached PR

closed 2023-10-05 by alexmucde

tomasdrozdskodaauto, 2023-11-02:
Thank you for the pull request. I will test it in the new release.