COVESA/dlt-viewer #303: "Copy selection to clipboard" timezone conversion to UTC

goezcan, 2022-12-06:
Copied lines contain the local time, i.e. adjusted to time zone where viewing computer is located. It is independent representation time set by Settings > Project Table > Time Zone settings. 

When those copied lines are pasted in a ticket and read by people living in other time zones, they are major source of confusion. Additional level of complexity is also added by daylight savings, daylight savings switch day difference between different countries and time zones having multiples of half hours difference to UTC. like Indian Standard Time (UTC+5:30)

Real world example to explain why it isn't a "champagne problem": Test drive performed in (UTC-5), carrying a data logger set for (UTC-7), analyzed in (UTC+1) and (UTC+5:30). First comment on the ticket is written few days before Europe ends summer time.

alexmucde, 2022-12-12:
I have verified, that there is an issue, that set timezone is not considered when copying lines. Must be fixed.

alexmucde, 2023-09-29:
@goezcan Found a solution, please check attached PR as a solution.
Fixies also all export functionalities like ASCII, CSV and Jira.

closed 2023-10-05 by alexmucde