COVESA/dlt-viewer #307: Sorting by Lifecycle, startup time

alexmucde, 2022-12-12:
Currently sorting by timestamp is only supported for a whole file, but not for dlt files containing several lifecycle.
Also startup time since the beginning of a lifecycle should be calculated and shown.

Feedback from Linkedin.

eaj40, 2024-03-23:
Can i work on this

alexmucde, 2024-03-25:
@eaj40 This would be great

eaj40, 2024-04-02:
Documentation for feature, page number in dlt viewer user manuual

eaj40, 2024-04-05:
Run dlt viewer on win11 pc and run dlt daemon on virtual box Ubuntu connect over TCP
Is this possible

alexmucde, 2024-04-08:
Yes this would help, then you can restart the VM for several Lifecycles.

eaj40, 2024-04-08:
Steps to reproduce the problem

How  to take traces for the problem

eaj40, 2024-04-26:
What do you mean by startup time

eaj40, 2024-05-02:
Solution ready, next steps