COVESA/dlt-viewer #315: how to connect with python

m17630030204, 2022-12-19:
hi:
i want to use python tcp  connect  ECU  and real time capture log, Do you have any suggestions for me!
thank

alexmucde, 2022-12-19:
@m17630030204 You can try the new DltTestRobot plugin. It provides a TCP connection. You will get the logs you are interested in as ASCII logs over the TCP connection. You have to send first the command filter add with EcuId, AppId and CtxId, so that you are getting the corresponding messages.

Please check the  DltTestRobot project for details and as a reference:
https://github.com/alexmucde/DLTTestRobot

I hope this is what you are searching for.

m17630030204, 2022-12-20:
hi 
I use tcp  client receive some dlt log data but dlt view can't loat it  
what do we need to do?  
such as:
=‰�ŠQNX��	 “cÄAQSYMINCVB���¨Œ����inc_vdata_service.45093�����info�B����A������0�Parse frame total number: 35890, changed: 16114
3D 89 00 8A 51 4E 58 00 00 09 A0 93 02 63 C4 08 41 06 51 53 59 4D 49 4E 43 56 42 00 00 00 A8 8C 00 02 00 00 18 00 69 6E 63 5F 76 64 61 74 61 5F 73 65 72 76 69 63 65 2E 34 35 30 39 33 00 00 02 00 00 05 00 69 6E 66 6F 00 42 00 00 00 08 00 41 00 00 00 05 00 02 00 00 30 00 50 61 72 73 65 20 66 72 61 6D 65 20 74 6F 74 61 6C 20 6E 75 6D 62 65 72 3A 20 33 35 38 39 30 2C 20 63 68 61 6E 67 65 64 3A 20 31 36 31 31 34 00

alexmucde, 2022-12-23:
DLT is in binary format. To directly connect to ECU you need a DLT decoder. If you use the DLTTestRobot Plugin, you will get everything in ASCII format.

alexmucde, 2023-06-02:
If there is still an issue or question please reopen the ticket, will close now.