COVESA/dlt-viewer #318: Failed to build on Ubuntu

lecherng, 2022-12-23:
Hi there,

Recently I tried to build dlt-viewer on my Ubuntu, and I noticed the following error:

> mainwindow.cpp: In member function ‘void MainWindow::initState()’:
mainwindow.cpp:262:37: error: ‘PlaceholderText’ is not a member of ‘QPalette’
         darkMode.setColor(QPalette::PlaceholderText, brightColor);
                                     ^~~~~~~~~~~~~~~
mainwindow.cpp:263:57: error: ‘PlaceholderText’ is not a member of ‘QPalette’
         darkMode.setColor(QPalette::Disabled, QPalette::PlaceholderText, disabledColor);
                                                         ^~~~~~~~~~~~~~~
mainwindow.cpp: In lambda function:
mainwindow.cpp:531:79: warning: unused parameter ‘paths’ [-Wunused-parameter]
     connect(this, &MainWindow::dltFileLoaded, this, [this](const QStringList& paths){
                                                            ~~~~~~~~~~~~~~~~~~~^~~~~

And I noticed the recent commit [https://github.com/COVESA/dlt-viewer/commit/04622f4097c832d3cb25d728f614280a51105a00](url) had removed the preprocessor for Q_OS_WIN. 
And now, to build for Linux, the requirement is no longer qt version 5.9.5, as mentioned in the INSTALL.md.

Is it possible to add the preprocessor for Q_OS_WIN or update the INSTALL.md to avoid confusion?

alexmucde, 2022-12-23:
@lecherng We have also identified that for Ubuntu 18.04 DLT Viewer does not build anymore. We do not officially support it anymore. If you have a fix, that it will also build with old versions, please provide a PR. Also for corrections in the documentation please provide a PR.

AlessandroBono, 2023-02-13:
@lecherng did you try to install it from [Flathub](https://beta.flathub.org/apps/org.genivi.DLTViewer)?