COVESA/dlt-viewer #327: With Qt6, payload with specific emoji, fails to render and make the line unreadable.

Elastino, 2023-01-13:
When DLT payload *contains* this emoji, ⚠️, Qt6 fails to render whole payload in Main tableview, Search view and DLT Plugin view. 

## Patches
Applying replace to ⚠️ to !(non-emoji) in _tablemodel.cpp:251_ and _searchtablemodel.cpp:166_, fixed the issue.
So far, only this emoji, ⚠️, affects display issue while other emoji likes 🚫 and ✅ displayed correctly.
It appears to be a problem in Qt6.4.2 but this information is shared as hot fix to the issue.
```
visu_data = msg.toStringPayload().trimmed().replace("⚠️","!");
```

## Before
[image]

## After
[image]

Elastino, 2023-01-13:
I forgot to mention that this is confirmed on Apple M1 with Ventura 13.1 and Qt6.4.2.

Elastino, 2023-01-16:
Confirmed to be a bug in Qt6.4.2 where **_unqualified_** emoji breaks whole line as unreadable.
I filed a QtBug, https://bugreports.qt.io/browse/QTBUG-110159

alexmucde, 2023-01-16:
@Elastino Thanks for the bug report. Can you also provide a PR, this will speed up the workaround?

Elastino, 2023-01-25:
Qt responded back that they need minimal example to reproduce the issue.
Minimal example is provided to Qt Bug ticket system, and progressed back to Reported stage.

https://bugreports.qt.io/browse/QTBUG-110159

alexmucde, 2023-05-16:
@Elastino Can you close the ticket?

Elastino, 2023-05-19:
This is not an issue of DLT-Viewer. Should be tracked in QTBUG for QT.

closed 2023-05-19 by Elastino