COVESA/dlt-viewer #334: File transfer plugin does not show received files

stoiantashkov, 2023-01-23:
I have an issue with DLT Filetransfer plugin. I can see the log in DLT viewer and also I see that the created package is transfered in the log but I can not see the file in Filetransfer plugin. DLT Viewer version is v2.19.0. DLT log is attached.

[filetransfer_failed_log.zip](https://github.com/COVESA/dlt-viewer/files/10477586/filetransfer_failed_log.zip)

alexmucde, 2023-01-25:
Please check the configuration file of the plugin. If you have a different context id like in the example folder of the plugin please change it:

<filetransferplugin_configuration>
<TAG_FLST>FLST</TAG_FLST>
<TAG_FLDA>FLDA</TAG_FLDA>
<TAG_FLFI>FLFI</TAG_FLFI>
<TAG_FLER>FLER</TAG_FLER>
<TAG_FLCTID>FILE</TAG_FLCTID>
</filetransferplugin_configuration>

There were also some changes in newer versions. AppId is now ignored.

alexmucde, 2023-05-16:
@stoiantashkov Is the issue solved? Please close ticket.