COVESA/dlt-viewer #335: DLT file transfer plugin not showing any files to save

GuruDlt, 2023-01-25:
[DLT_crash.zip](https://github.com/COVESA/dlt-viewer/files/10497934/DLT_crash.zip)
[Dlt-viewer.zip](https://github.com/COVESA/dlt-viewer/files/10498003/Dlt-viewer.zip)

Hi Team,

I have dlt file transfer code to transfer core binary to dlt viewer and I am able to see logs also from dlt-viewer(as attached zip[DLT_crash.zip]) but I don't see any files to save from file transfer plugin as shown in screen shot from Dlt-viewer.zip.

Could anyone please help me here with what I am missing.

Thanks in advance.

Guru.

alexmucde, 2023-01-25:
Please check the configuration file of the plugin. If you have a different context id like in the example folder of the plugin please change it:
<?xml version="1.0" encoding="UTF-8"?>
<filetransferplugin_configuration>
<TAG_FLST>FLST</TAG_FLST>
<TAG_FLDA>FLDA</TAG_FLDA>     
<TAG_FLFI>FLFI</TAG_FLFI>     
<TAG_FLER>FLER</TAG_FLER>     
<TAG_FLCTID>FILE</TAG_FLCTID>   
 </filetransferplugin_configuration>

GuruDlt, 2023-01-25:
Hi Alex

From dlt open source, I don't see file transfer plugin config file. please refer below
~/Documents/dlt-daemon-master$ find -iname "*file*"
./include/dlt/dlt_filetransfer.h
./doc/dlt_filetransfer.md
./tests/testfile_filetransfer.txt
./tests/mod_system_logger/Makefile
./tests/start_filetransfer_test.sh
./tests/testfile_extended.dlt
./tests/testfile.dlt
./gtest-1.7.0/include/gtest/internal/gtest-filepath.h
./gtest-1.7.0/Makefile.am
./gtest-1.7.0/scripts/fuse_gtest_files.py
./gtest-1.7.0/scripts/test/Makefile
./gtest-1.7.0/Makefile.in
./gtest-1.7.0/src/gtest-filepath.cc
./gtest-1.7.0/make/Makefile
./gtest-1.7.0/test/gtest-filepath_test.cc
./gtest-1.7.0/test/gtest_xml_outfile2_test_.cc
./gtest-1.7.0/test/gtest_xml_outfiles_test.py
./gtest-1.7.0/test/gtest_xml_outfile1_test_.cc
./gtest-1.7.0/Makefile
./src/lib/dlt_filetransfer.c
./src/examples/dlt-example-filetransfer.c
./src/tests/dlt-test-receiver-multiple-files.sh
./src/tests/dlt-test-filetransfer.c
./src/tests/dlt-test-filetransfer-file
./src/tests/dlt-test-filetransfer-image.png
./src/system/dlt-system-logfile.c
./src/system/dlt-system-filetransfer.c
./src/shared/dlt_config_file_parser.h
./src/shared/dlt_config_file_parser.c
./build/.cmake/api/v1/query/client-integration-vscode/cmakeFiles-v1
./build/CMakeFiles
./distfiles
~/Documents/dlt-daemon-master$

alexmucde, 2023-01-25:
Look here: https://github.com/COVESA/dlt-viewer/blob/master/plugin/examples/filetransferplugin_configuration.xml
If you adopt this file and change it to your configuration it should work.

GuruDlt, 2023-01-30:
Hi Alex,

Sorry for the bit delayed reply here. I have understood that this configuration xml need some changes as per my configuration.
As below is the content of xml config file, can you please share any reference / example on how or what basis changes should be done in xml config file.

<?xml version="1.0" encoding="UTF-8"?>
<filetransferplugin_configuration>
<TAG_FLST>FLST</TAG_FLST>
<TAG_FLDA>FLDA</TAG_FLDA>
<TAG_FLFI>FLFI</TAG_FLFI>
<TAG_FLER>FLER</TAG_FLER>
<TAG_FLCTID>FILE</TAG_FLCTID>
</filetransferplugin_configuration>

alexmucde, 2023-01-30:
@GuruDlt You have to set the TAG_FLCTID to your used CtxId or change your SW to use CtxId FILE. Then you have to load this file into the Plugin.

GuruDlt, 2023-01-31:
Hi @alexmucde 

Thanks for input and it worked when I changed context id to use the same by plugin from sw. 
And actually I am transferring core file when it generated and i have saved core file from dlt viewer and pulled out core from target to PC and compared both core files, but when I compared with beyond compare(binary comparision), the content is not the same. **But number of bytes are same.**

Any idea why and what could be wrong.

GuruDlt, 2023-01-31:
Hi @alexmucde 

After I changed in sw to use same context id as per plugin configuration file, it worked and now I am able to save the file from dlt application.

Thanks for inputs @alexmucde. I am closing the ticket with this ticket.

closed 2023-01-31 by GuruDlt