COVESA/dlt-viewer #339: dlt file transfer incomplete

GuruDlt, 2023-02-02:
[Incomplete_dlt.zip](https://github.com/COVESA/dlt-viewer/files/10568184/Incomplete_dlt.zip)

Hi Team

as per attached logs and screen shot when I am transferring file with dlt file transfer plug in, its transferred incomplete always.
Below is my conf:
FiletransferEnable = 1
FiletransferDirectory = /dev/shmem/
FiletransferCompression = 1
FiletransferCompressionLevel = 5

could you please help me.

alexmucde, 2023-02-03:
@GuruDlt Must be a problem with your generator.

GuruDlt, 2023-02-03:
> 
Hi @alexmucde 
are you saying some issue with SW?

GuruDlt, 2023-02-03:
Hi @alexmucde 

are you saying the issue with SW?

GuruDlt, 2023-05-24:
@alexmucde 
yes its not because of SW issue but

-> it can happen because of when DLT was loaded, in this case still can expect improvements from dlt.
-> it can happen when if file transfer starts before file data copied into file

closed 2023-05-24 by GuruDlt