COVESA/dlt-viewer #344: How to apply parser to decode logfiles (Non Verbose Mode Plugin) buy command line

Znzero, 2023-02-15:
[image]
i have many logfiles in nonverbose mode, how do i use command line to decode those logfiles?

alexmucde, 2023-05-16:
@Znzero

Option1: There is currently no command line interface for the non verbose plugin to load a configuration. But you can create and save a project, where you have enabled the plugin and you have loaded a Fibex file. Then load this project file by commandline.

Option2: The plugin must be extended to support the QDltPluginCommandInterface, then it would be also possible to provide the fibex file by commandline.

Can you work with Option1 or should we extend the plugin to support also Option2?

alexmucde, 2023-06-02:
@Znzero If your question is answered please close the ticket. If you tend to option 2, we can keep it open.

Znzero, 2023-06-02:
> @Znzero If your question is answered please close the ticket. If you tend to option 2, we can keep it open.
my question is answered, thanks~~

closed 2023-06-02 by Znzero