COVESA/dlt-viewer #367: Can not build on MacOS from Release v2.24.0

SangTruongTan, 2023-07-23:
Hi,
I've just faced a problem when I tried to build the app for my MacOS. I think that there is a change between v2.23.0 and v2.24.0 (later on). It got the following error and I can by pass it by static case the variable.
Error:
```
Project/dlt-viewer/qdlt/qdltconnection.cpp:79:12: error: no viable conversion from 'QDltDataView' to 'QByteArray'
    data = dataView + bytes;
```
Remedy:
```
diff --git a/qdlt/qdltconnection.cpp b/qdlt/qdltconnection.cpp
index 20277fa..2534105 100644
--- a/qdlt/qdltconnection.cpp
+++ b/qdlt/qdltconnection.cpp
@@ -75,7 +75,7 @@ void QDltConnection::add(const QByteArray &bytes)
 {
     bytesReceived += bytes.size();
 
-    data = dataView + bytes;
+    data = static_cast<QByteArray>(dataView) + bytes;
```

Hit2theMo, 2023-08-07:
Thanks a lot! This solved the same issue I faced.

hannesa2, 2023-08-11:
Where exactly is you issue. I just created a new version https://github.com/COVESA/dlt-viewer/releases/tag/v2.25.0 and it builds for macOS, at least for x86

SangTruongTan, 2023-08-11:
I'm using Apple Silicon, and it's also experienced on v2.25.0 again with the same point.

hannesa2, 2023-08-12:
I applied this change in #369
[image]

But locally on by M1 I still run into 
[image]

Even on CI https://github.com/COVESA/dlt-viewer/actions/runs/5839431094/job/15837605503?pr=369 it runs into the same error
[image]

@SangTruongTan 
Please can you make a pull request to demonstrate, what exactly needs to be done ?

SangTruongTan, 2023-08-12:
Hi @hannesa2,
It's very strange, that's the only change at my end. Did you try flushing cmake cache?

hannesa2, 2023-08-12:
I did a `git clean -xdf`
What's your recommendation to flush cmake cache ? Maybe you should do this as well.

SangTruongTan, 2023-08-12:
I believe it's a clean version I have just done it so far. The remedy is also working with @Hit2theMo.

alexmucde, 2023-08-17:
Reopen as fix is not working.

hannesa2, 2023-08-17:
> Please check at your end.

There is nothing to check on my send. Simple look at CI #369 and the resulting error

SangTruongTan, 2023-08-20:
Fixed by the remedy: https://github.com/COVESA/dlt-viewer/pull/373/commits/4bb3b26802316cd3fa28fde92f1ea32a72070fa3

closed 2023-08-20 by SangTruongTan