COVESA/dlt-viewer #376: Zero Termination displayed as space

smarkwardt, 2023-09-08:
Wrong Space in decoded Payload when an argument of type string is followed by an argument of an other type.
[image]

alexmucde, 2023-09-18:
@smarkwardt Please tell me to which plugin etc. this issue is related? Is this an open source plugin?

smarkwardt, 2023-09-18:
@alexmucde 
I think it's a basic functionality of the dlt-viewer and not a plugin.

The spaces are incorrectly displayed by the dlt viewer.

````
DLT_LOG(mycontext, DLT_LOG_INFO, DLT_STRING("val: '"), DLT_INT(1), DLT_STRING("'"));
DLT_LOG(mycontext, DLT_LOG_INFO, DLT_STRING("aaa"), DLT_STRING("bbb"));
````

[image]

From my point of view, it is unfavorable because you have to take the unnecessary spaces in the filter query into account.

alexmucde, 2023-09-18:
Thanks for the detailed explanation. Now i understand better your issue.
Arguments of a DLT messages are separated by a space, when converted to ASCII.
This feature is implemented since the beginning of the DLT Viewer development.
The users of the DLT Viewer are used to this feature.
Other logging frameworks like Qt Logging behave the same.

closed 2023-09-18 by smarkwardt