COVESA/dlt-viewer #380: Build arm64 images

hannesa2, 2023-10-02:
Now we could build artifacts for Apple silicon as well, because M1 runner are available now https://github.com/actions/runner-images/issues/8439
But it makes no sense, because we have no sponsor for an Apple developer account. With this, an arm64 artifact for MacOS would make sense.
If you think, you could be the sponsor, please ping me.

radeva, 2024-01-18:
@hannesa2 You can try [FlyCI](https://www.flyci.net/)'s M1 and M2 runners. They are on average 2x faster and 2x cheaper than GitHub's **AND** we have a **free tier for OSS projects** (see below).

# Install Instructions

1. Install [FlyCI app](https://github.com/apps/flyci-prod) and 
2. Easily replace one line of code and start using FlyCI runners:

```diff
jobs:
 ci:
-    runs-on: macos-latest
+    runs-on: flyci-macos-large-latest-m1
   steps:
   - name: 👀 Checkout repo
     uses: actions/checkout@v4
```
# 500 mins/month Free for Public Repos

Since your repo is public, FlyCI offers **500 mins/month of free M1 runner usage** with the `flyci-macos-large-latest-m1` runner for public projects. 

Don't hesitate to contact us in case the free tier doesn't suit your needs or you experience any issues with the runners. Our team is here to support you!

Best Regards,
Veselina Radeva
Product Manager at [FlyCI](https://flyci.net/)

hannesa2, 2024-01-18:
Thank you for the hint, it worth to have a look. 
Currently I do my experiments in my fork with an own custom arm64 runner https://github.com/hannesa2/dlt-viewer/pulls because I can't access setting in this fork

hannesa2, 2024-02-25:
@alexmucde 
[image]

You label it with "help wanted" but the only one who can help here, are you !
I can't access "Settings" tab with runner settings, and only you can do this or provide access to me.

bunty95, 2024-06-13:
@hannesa2 issue can be closed?

hannesa2, 2024-06-13:
The signing is missing. The idea is to do it analog to https://github.com/gitx/gitx/pull/334/files
But maybe we should open an new pull request for this

hannesa2, 2024-06-13:
> Currently I do my experiments in my fork with an own custom arm64 runner https://github.com/hannesa2/dlt-viewer/pulls because I can't access setting in this fork

This is solved be cause macos-14 is a free arm64 runner