COVESA/dlt-viewer #382: QT build on MacOS

bunty95, 2023-10-03:
Using below commands, the DLT project no longer creates a "DLT-Viewer.app" file.
cMake worked up until 2.21.1 but no longer works.

Using the below qmake commands, a ".app" isn't created.

brew install qt@5
mkdir build && cd build
/usr/local/opt/qt@5/bin/qmake ../BuildDltViewer.pro
make

- Creates a "Unix Executable" rather then a .app file.

Any help would be greatly appreciated

hannesa2, 2023-10-03:
Here https://github.com/COVESA/dlt-viewer/blob/master/.github/workflows/BuildPR.yml#L28-L40 you see how the CI builds it on MacOS. But it's just x86

bunty95, 2023-10-04:
The build works correctly for me as well. The problem that I have is that a ".APP" file isn't created. It used to work with cMake up until 2.21.1, but after that broke. 

Trying to ask for help on creating a .APP file again.

bunty95, 2023-12-10:
Update: when using the build.sh adding the following command resolve's the issue.

sudo xcode-select -s /Applications/Xcode.app

closed 2023-12-10 by bunty95