COVESA/dlt-viewer #383: How to retrieve all "dlt" files by dlt-viewer

YongJ7, 2023-10-05:
Hello, 

I wonder if there is any way to retrieve all saved "dlt" format log files by using DLT-viewer.

I was expecting Filetransfer Plugin would do, but it seems it only works with binary files.

Please let me know if this is possible.

Best regards.

alexmucde, 2023-10-05:
Hi @YongJ7,

i do not understand your issue. Can you please explain in more detail, what you want to achieve?

YongJ7, 2023-10-05:
Dear @alexmucde,

As far as I know, when extracting logs through dlt-viewer, only the logs currently visible in the viewer can be saved as a file. 

I am wondering if it is possible to save all dlt logs which are in the storage using a plugin, or something else.

Thanks!

alexmucde, 2023-10-05:
You can use the command line options to extract the files using DLT Viewer.

Two possibilities to extract Code Dump files automatically:

1) Option -e “Filetransfer Plugin|export|C:\tmp“

2) Add "<AUTOSAVE>c:\tmp</AUTOSAVE>" to the configuration file of the plugin.

YongJ7, 2023-10-05:
Dear @alexmucde,

I'm sorry but could you please explain a little more specifically? 

I added using method 2, but nothing seems to happen. What am I missing?

Thanks.

alexmucde, 2023-10-05:
@YongJ7 You have to add the Autosave option to the configuration file of the Filetransfer Plugin, you have to load the configuration file in the plugin. I hope you are familiar with this. If you are working in a BMW project i can help you and we can setup a meeting.

YongJ7, 2023-10-10:
Dear @alexmucde,
I'm trying to test in Linux PC.
I have added at .xml file like below, and started dlt-daemon & dlt-system.
Then set plugin in dlt-viewer like second image.
And nothing happened,,,
I was expecting files saved in dlt-viewer's Filetransfer Plugin.
What did I do wrong? And I'm not working on a BMW project.
[image]
[image]

Thank You

alexmucde, 2023-10-10:
@YongJ7 Can you see the files in the Filetransfer Plugin? If they are not visible and complete "green background", the appid and ctxid could be wrong.

YongJ7, 2023-10-11:
Dear @alexmucde,
I'm not sure but I think it only works with via [ sudo dlt-viewer ]. 
Maybe there is a permission problem or something in my PC. 
Any way It worked! Thanks a lot. I'll  close the issue.

closed 2023-10-11 by YongJ7