COVESA/dlt-viewer #391: Copy logs works only 1 selected line

kimmmmk, 2023-11-09:
I selected multi lines on table and copy them to clipboard, It just copied one line.
I tested Ctrl+C and menu (Copy selection to Clipboard)

Here is my DLT viewer's version information. 

---------------------------
Package Version : 2.25.0 
Package Revision: unstable

Build Date: Nov  9 2023
Build Time: 18:10:06
Qt Version: 6.2.10
Compiler: msvc
Compiler Version: 1929

Architecture: 64 Bit
Architecture: Little Endian

(C) 2016,2022 BMW AG

---------------------------

Here is Application output from Qt creator 

Start DLT export of 9 messages ,silent mode false
Exported: 0.00 %
100 % DLT export done for 9 messages with result 0

kimmmmk, 2023-11-09:
Here is my clipboard after ctrl+c
14 2020/09/20 19:47:15.217104 198.0273 87 ECU1 SYS JOUR 896 log error verbose 3 kernel:

Here is debug information clipbardString during copy tables.

I think the "\0" is problem. 

Locals		
		buf	"DLT\00133g_fP\003\000ECU1=X\000\ECU1\000\000\003\200\000\0369ÃA\003SYS\000JOUR\000\002\000\000\026\000systemd-journal[467]:\000\000\002\000\000\001\000\000\000\002\000\000\031\000Missed 4 kernel m"... (108)	QByteArray
		msg	@0x1ecb4f8c30	QDltMsg
		num	1	unsigned long
		this	@0x1ecb4f8e40	DltExporter
	Inspector		
	Expressions		
		clipboardString	"14 2020/09/20 19:47:15.217104 198.0273 87 ECU1 SYS JOUR 896 log error verbose 3 kernel:\000 \000 mttcan c3"... (542)	QString
				'1' 	49      	0x0031	QChar
				'4' 	52      	0x0034	QChar
				' ' 	32      	0x0020	QChar
				'2' 	50      	0x0032	QChar
				'0' 	48      	0x0030	QChar
				'2' 	50      	0x0032	QChar
				'0' 	48      	0x0030	QChar
				'/' 	47      	0x002f	QChar
				'0' 	48      	0x0030	QChar
				'9' 	57      	0x0039	QChar
				'/' 	47      	0x002f	QChar
				'2' 	50      	0x0032	QChar
				'0' 	48      	0x0030	QChar
				' ' 	32      	0x0020	QChar
				'1' 	49      	0x0031	QChar
				'9' 	57      	0x0039	QChar
				':' 	58      	0x003a	QChar
				'4' 	52      	0x0034	QChar
				'7' 	55      	0x0037	QChar
				':' 	58      	0x003a	QChar
				'1' 	49      	0x0031	QChar
				'5' 	53      	0x0035	QChar
				'.' 	46      	0x002e	QChar
				'2' 	50      	0x0032	QChar
				'1' 	49      	0x0031	QChar
				'7' 	55      	0x0037	QChar
				'1' 	49      	0x0031	QChar
				'0' 	48      	0x0030	QChar
				'4' 	52      	0x0034	QChar
				' ' 	32      	0x0020	QChar
				'1' 	49      	0x0031	QChar
				'9' 	57      	0x0039	QChar
				'8' 	56      	0x0038	QChar
				'.' 	46      	0x002e	QChar
				'0' 	48      	0x0030	QChar
				'2' 	50      	0x0032	QChar
				'7' 	55      	0x0037	QChar
				'3' 	51      	0x0033	QChar
				' ' 	32      	0x0020	QChar
				'8' 	56      	0x0038	QChar
				'7' 	55      	0x0037	QChar
				' ' 	32      	0x0020	QChar
				'E' 	69      	0x0045	QChar
				'C' 	67      	0x0043	QChar
				'U' 	85      	0x0055	QChar
				'1' 	49      	0x0031	QChar
				' ' 	32      	0x0020	QChar
				'S' 	83      	0x0053	QChar
				'Y' 	89      	0x0059	QChar
				'S' 	83      	0x0053	QChar
				' ' 	32      	0x0020	QChar
				'J' 	74      	0x004a	QChar
				'O' 	79      	0x004f	QChar
				'U' 	85      	0x0055	QChar
				'R' 	82      	0x0052	QChar
				' ' 	32      	0x0020	QChar
				'8' 	56      	0x0038	QChar
				'9' 	57      	0x0039	QChar
				'6' 	54      	0x0036	QChar
				' ' 	32      	0x0020	QChar
				'l' 	108      	0x006c	QChar
				'o' 	111      	0x006f	QChar
				'g' 	103      	0x0067	QChar
				' ' 	32      	0x0020	QChar
				'e' 	101      	0x0065	QChar
				'r' 	114      	0x0072	QChar
				'r' 	114      	0x0072	QChar
				'o' 	111      	0x006f	QChar
				'r' 	114      	0x0072	QChar
				' ' 	32      	0x0020	QChar
				'v' 	118      	0x0076	QChar
				'e' 	101      	0x0065	QChar
				'r' 	114      	0x0072	QChar
				'b' 	98      	0x0062	QChar
				'o' 	111      	0x006f	QChar
				's' 	115      	0x0073	QChar
				'e' 	101      	0x0065	QChar
				' ' 	32      	0x0020	QChar
				'3' 	51      	0x0033	QChar
				' ' 	32      	0x0020	QChar
				'k' 	107      	0x006b	QChar
				'e' 	101      	0x0065	QChar
				'r' 	114      	0x0072	QChar
				'n' 	110      	0x006e	QChar
				'e' 	101      	0x0065	QChar
				'l' 	108      	0x006c	QChar
				':' 	58      	0x003a	QChar
				'\0'	0      	0x0000	QChar
				' ' 	32      	0x0020	QChar
				'\0'	0      	0x0000	QChar

alexmucde, 2023-12-06:
@kimmmmk Thanks for the bug report and the problem analysis. Perhaps we have to remove all "\0" characters in the export to clipboard function.

alexmucde, 2024-02-23:
@kimmmmk Please check the attached PR, if this solves the issue.

closed 2024-02-26 by alexmucde