COVESA/dlt-viewer #404: Apple M1 failed to run version v2.25.0

Kayson-1994, 2023-12-27:
Process:               dlt-viewer [34319]
Path:                  /private/var/folders/*/DLTViewer.app/Contents/MacOS/dlt-viewer
Identifier:            org.genivi.dlt-viewer
Version:               2.25.0 (2.25.0)
Code Type:             X86-64 (Translated)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-12-27 13:16:46.9208 +0800
OS Version:            macOS 14.1.1 (23B81)
Report Version:        12

System Integrity Protection: enabled

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/qt@5/lib/QtNetwork.framework/Versions/5/QtNetwork
Referenced from: <AB834FDA-69A2-35B3-9110-EF1DE454810F> /Users/USER/*/libqdlt.dylib
Reason: tried: '/usr/local/opt/qt@5/lib/QtNetwork.framework/Versions/5/QtNetwork' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/qt@5/lib/QtNetwork.framework/Versions/5/QtNetwork' (no such file), '/usr/local/opt/qt@5/lib/QtNetwork.framework/Versions/5/QtNetwork' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/Library/Frameworks/QtNetwork.framework/Versions/5/QtNetwork' (no such file), '/System/Library/Frameworks/QtNetwork.framework/Versions/5/QtNetwork' (no such file, not in dyld cache), '/opt/homebrew/Cellar/qt@5/5.15.11/lib/QtNetwork.framework/Versions/5/QtNetwork' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64
(terminated at launch; ignore backtrace)

hannesa2, 2023-12-27:
> '/usr/local/opt/qt@5/lib/QtNetwork.framework/Versions/5/QtNetwork' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/qt@5

Not sure, but maybe you have a x86_54 qt@5 ?

15031619, 2024-01-14:
dear developers,

my laptop has same problem, but the reason is different: tried: '$ORIGIN/../lib/libqdlt.dylib' (no such file)....

could you give me some suggestion?

thank you from my heart.

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: @rpath/libqdlt.dylib
Referenced from: <E5A41B71-7EB9-3404-A4D9-4475992D35CC> /private/var/folders/*/DLTViewer.app/Contents/MacOS/dlt-viewer
Reason: tried: '$ORIGIN/../lib/libqdlt.dylib' (no such file), '/usr/local/lib/QtCore.framework/libqdlt.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/QtCore.framework/libqdlt.dylib' (no such file), '$ORIGIN/../lib/libqdlt.dylib' (no such file), '/usr/local/lib/QtCore.framework/libqdlt.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/QtCore.framework/libqdlt.dylib' (no such file), '/usr/local/lib/libqdlt.dylib' (no such file), '/usr/lib/libqdlt.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

hannesa2, 2024-01-25:
https://github.com/COVESA/dlt-viewer/issues/398#issuecomment-1910528420

15031619, 2024-01-27:
> [#398 (comment)](https://github.com/COVESA/dlt-viewer/issues/398#issuecomment-1910528420)

thanks ,it works

alexmucde, 2024-03-04:
@Kayson-1994 @15031619 Please close ticket if issue is solved.

closed 2024-09-09 by alexmucde