COVESA/dlt-viewer #414: Apple M1 MAC Failure: no such file or directory

saurav2502, 2024-01-25:
2 warnings generated.
[105/105] Linking CXX executable bin/dlt-viewer
+ cpack -G External
CPack: Create package using External
CPack: Install projects
CPack: - Install project: dlt-viewer []
CPack: Create package
-- Install configuration: "Release"
-- Installing: /Users/saurav/dlt-viewer/build/install/DLTViewer.app/Contents/Frameworks/libqdlt.dylib
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/lib/libqdlt.dylib
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/.
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdlt.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltcontrol.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdlttcpconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltfile.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltpluginmanager.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./export_rules.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltmsg.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltsegmentedmsg.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./dlt_user.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltserialconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./dlt_common_cfg.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltsettingsmanager.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltfilter.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./dlt_protocol.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./plugininterface.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltbase.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltfilterlist.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltmessagedecoder.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltplugin.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltargument.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltoptmanager.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltudpconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltdefaultfilter.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltipconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltfilterindex.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./dlt_types.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./dlt_common.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./sdk/include/qdlt/./qdltconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/.
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdlt.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltcontrol.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdlttcpconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltfile.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltpluginmanager.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./export_rules.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltmsg.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltsegmentedmsg.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./dlt_user.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltserialconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./dlt_common_cfg.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltsettingsmanager.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltfilter.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./dlt_protocol.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./plugininterface.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltbase.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltfilterlist.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltmessagedecoder.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltplugin.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltargument.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltoptmanager.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltudpconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltdefaultfilter.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltipconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltfilterindex.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./dlt_types.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./dlt_common.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./include/qdlt/./qdltconnection.h
-- Installing: /Users/saurav/dlt-viewer/build/install/./src/dummycommandplugin
-- Installing: /Users/saurav/dlt-viewer/build/install/./src/dummycommandplugin/CMakeLists.txt
[... 80 lines cut ...]

# execute_process(COMMAND tree
#     WORKING_DIRECTORY "${CMAKE_BINARY_DIR}"
#     RESULT_VARIABLE STATUS)

file(GLOB DLT_PLUGINS_SO "${CMAKE_INSTALL_PREFIX}/${DLT_PLUGIN_INSTALLATION_PATH}/*.so")
list(TRANSFORM DLT_PLUGINS_SO PREPEND "-executable=")

message(STATUS "Call ${MACDEPLOYQT_EXECUTABLE} ${DLT_APP_DIR_NAME} ${DLT_PLUGINS_SO}")
execute_process(COMMAND
    "${MACDEPLOYQT_EXECUTABLE}"
    "${DLT_APP_DIR_NAME}"
    -verbose=1
    -always-overwrite
    -dmg
    -libpath=${CMAKE_INSTALL_PREFIX}/${DLT_APP_DIR_NAME}/Contents/Frameworks
    ${DLT_PLUGINS_SO}
    WORKING_DIRECTORY "${CMAKE_INSTALL_PREFIX}"
    RESULT_VARIABLE STATUS)

if(STATUS AND NOT STATUS EQUAL 0)
    message(SEND_ERROR "Failure: ${STATUS}")
else()
    message(STATUS "Success.")
endif()

hannesa2, 2024-01-25:
I made this work on my fork https://github.com/dlt-viewer/dlt-viewer/pull/50
It's not on upstream (here) because I'll wait for the permission to add a custom M1 runner here. @alexmucde

hannesa2, 2024-01-25:
Btw, the magic was this https://github.com/dlt-viewer/dlt-viewer/pull/50/files#diff-1a6b54f881d5b938ae705efdae5a93ec8f3eac68ad15e44bf962afd94c2db707R10-R15

saurav2502, 2024-01-25:
Thanks for quick information.

it was build successfully and it created binary 
cd ~/dlt-viewer/build/bin 
ls -al
drwxr-xr-x   5 saurav  603018132      160 Jan 25 22:37 .
drwxr-xr-x  26 saurav  603018132      832 Jan 25 22:45 ..
-rwxr-xr-x   1 saurav  603018132  1349496 Jan 25 22:37 dlt-viewer
-rwxr-xr-x   1 saurav  603018132   347888 Jan 25 22:37 libqdlt.dylib
drwxr-xr-x  10 saurav  603018132      320 Jan 25 22:37 plugins

When I am running using command `./dlt-viewer` then dot-viewer is launching but it is very unstable, getting closed immediately.
[1]    28067 segmentation fault  ./dlt-viewer
[image]
[image]

saurav2502, 2024-01-25:
Another binary present in 
/dlt-viewer/build/dist

 DLTViewer.app  
 DLTViewer.dmg  
 LICENSE.txt  
 MPL.txt  
 README.md  
 ReleaseNotes_Viewer.txt  
 doc  
filters  
include  
plugins  
sdk  
src

# how to launch DLTViewer.app?
I tried `open -a DLTViewer.app` > did not work.
double click on .dmg file > it just open below tab but does nothing.
[image]
Can you help me here that how this could be resolved the launching of the dot-viewer? It would be highly appreciated

bunty95, 2024-01-26:
> Btw, the magic was this https://github.com/dlt-viewer/dlt-viewer/pull/50/files#diff-1a6b54f881d5b938ae705efdae5a93ec8f3eac68ad15e44bf962afd94c2db707R10-R15

This doesn't seem to work for non M1 Mac devices. When trying on Intel Mac's CPACK still fails.

closed 2024-06-13 by hannesa2