COVESA/dlt-viewer #416: Improvement - Add multilanguage GUI

bovirus, 2024-01-29:
@alexmucde 

Please consider to modify source code to have multilanguage GUI.

With Qt could be easier.

bovirus, 2024-02-24:
@alexmucde 

About translation I can help with Italian language.

vifactor, 2025-02-28:
I used to develop multilanguage desktop apps for many years and I must say internationalization requires insane amount of work, definitely much more than this project has resources available.

alexmucde, 2025-07-24:
Currently no resources available to support multilanguage.

closed 2025-07-24 by alexmucde

bovirus, 2025-07-26:
@vifactor

Depend which development language are you using to create the application.

There some development languages/tools that are think to create multilanguage app.

One example is QT SDK (that use .ts/.qm language file).