COVESA/dlt-viewer #420: Qt6: Filtransfer Plugin does not find any files naymore

alexmucde, 2024-02-01:
Filetransfer does not show any found  files anymore

minminlittleshrimp, 2024-02-01:
Hello @alexmucde 
I have no idea if my issue the same. I try to send a file with dlt-example-filetransfer, although the binary code can be seen, filetransfer plugin show nothing:
```
2 2024/02/01 19:34:06.296903 21546.4870 0 ECU1 FLTR FLTR 1561438 log info verbose 8 FLST 1624498309 /usr/local/share/dlt-filetransfer/dlt-test-filetransfer-file 17 Thu Feb 1 19:14:05 2024 1 1024 FLST
3 2024/02/01 19:34:06.296913 21546.4870 1 ECU1 FLTR FLTR 1561438 log info verbose 5 FLDA 1624498309 1 21 21 21 48 65 6c 6c 6f 57 6f 72 6c 64 21 21 21 0a FLDA
5 2024/02/01 19:34:06.337900 21546.4887 2 ECU1 FLTR FLTR 1561438 log info verbose 3 FLFI 1624498309 FLFI
426 2024/02/01 19:35:43.952592 21644.1427 0 ECU1 FLTR FLTR 1561521 log info verbose 8 FLST 2202457823 /usr/local/share/dlt-filetransfer/dlt-test-filetransfer-image.png 18897 Thu Feb 1 19:14:05 2024 19 1024 FLST
427 2024/02/01 19:35:43.952603 21644.1427 1 ECU1 FLTR FLTR 1561521 log info verbose 5 FLDA 2202457823 1 89 50 4e 47 0a 1a 0a 00 00 00 0d 49 48 44 52 00 00 02 ac 00 00 01 1b 08 06 00 00 00 1f e5 d6 60 00 00 00 01 73 52 47 42 00 ae ce 1c e9 00 00 00 04 67 41 4d 41 00 00 b1 8f 0b fc 61 05 00 00 00 09 70 48 59 73 00 00 0e c3 00 00 0e c3 01 c7 6f a8 64 00 00 49 6a 49 44 41 54 78 5e ed 9d f9 fb 15 c5 9d ef fd b3 e6 ce 7d 66 b9 db 2f 73 27 93 67 ee 24 93 3b d9 e6 66 37 8b 99 98 68 26 8b a3 26 71 c3 48 d4 88 1b 2a 2a 8a 80 c8 22 fb 8e ec 9b 28 28 2a a8 6c a2 28 2e 08 88 b8 20 a9 5b ef 36 ed b4 6d 6d dd a7 cf 39 75 be e7 c5 f3 f4 a3 d0 dd b5 bc 3f af aa 7a 57 75 75 9f 0b 0c 7f 50 00 05 50 00 05 50 00 05 50 00 05 50 20 63 05 2e c8 b8 6c 14 0d 05 50 00 05 50 00 05 50 00 05 50 00 05 0c 86 15 08 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 00 c3 9a 75 78 28 1c 0a a0 00 0a a0 00 0a a0 00 0a a0 00 86 15 06 50 00 05 50 00 05 50 00 05 50 00 05 b2 56 60 6c 0d eb 9f fe f4 27 53 1e e7 cf 9f 37 1c 68 00 03 30 00 03 30 00 03 30 30 6c 06 aa fe 44 ff cf 9f 8f 15 18 1b c3 5a 35 a7 67 df ff d0 3c f9 e2 1b 66 cd 93 2f 9b fb 57 ed 37 57 3c f8 a4 b9 e4 ee 1d 9f 1c ff 6f f2 46 f3 77 97 af e2 40 03 18 80 01 18 80 01 18 80 81 be 31 f0 af bf df f8 29 ff f1 8b fb 1e 2f 7c c9 f2 9d 2f 15 3e e5 d4 99 f7 8b 05 b5 d2 c3 8c b3 79 9d d0 86 b5 6a 52 4f be f3 9e 59 b6 e3 88 b9 6a e6 6e f3 85 ab d7 9a 9f de bd d3 5c 39 e3 69 73 fb d2 03 66 f1 8e e3 66 dd 53 27 3e 39 f6 1c 3e 6b 5e 39 61 38 d0 00 06 60 00 06 60 00 06 60 a0 6f 0c c8 6f 54 fd c7 8a 27 de 2c 7c c9 55 b3 f6 16 3e e5 0b 57 ad 33 32 b1 8f 6e 39 64 5e 7f fb 5d f3 d1 47 1f 7d 62 60 c7 cd bc 4e 38 c3 5a 35 a9 0a ae 82 ac 60 2b e8 02 60 de d6 57 cd a1 37 ce 9b 97 6d 03 e4 40 03 18 80 01 18 80 01 18 80 81 9c 19 90 89 bd fe 91 e7 cc 97 27 ad 37 df bf 65 8b 99 b3 e1 80 39 fc da a9 c2 bc 8e d3 ca eb 84 31 ac 65 d0 14 c0 7d 2f bd 55 98 54 05 57 41 56 b0 73 86 91 b2 d1 59 c2 00 0c c0 00 0c c0 00 0c c4 18 d8 ba ff b4 f9 c3 fc 7d e6 5b 37 6e 2e cc eb 13 2f bc 3e 36 ab ae 13 c2 b0 ca ac ca a8 1e 7d fd 94 b9 fe e1 a7 cc f7 a6 6c 35 2b 76 59 93 fa 96 85 9f 03 0d 60 00 06 60 00 06 60 00 06 26 18 03 5b f7 9d 2e b6 0d 68 81 4e 0b 75 13 7d c5 75 a4 0d ab 8c aa 36 23 bf 7d fa ac 99 b2 e8 59 f3 cd 1b b7 98 87 37 be 42 a3 9c 60 8d 92 49 07 13 2f 18 80 01 18 80 01 18 70 33 a0 05 FLDA
429 2024/02/01 19:35:43.993851 21644.1445 2 ECU1 FLTR FLTR 1561521 log info verbose 5 FLDA 2202457823 2 3a 2d d4 69 c1 ee f8 89 33 9f 18 d7 89 b6 c7 75 24 0d 6b 69 54 cf 9d 3b 67 ee 5b f9 bc f9 d7 c9 9b cc bd ab 0e 9b 83 af 9f 37 47 ad 59 e3 40 03 18 80 01 18 80 01 18 80 81 71 62 40 0b 76 5f b7 5f 39 9a ba e4 59 73 e6 ec 7f 7e 5d 60 a2 18 d7 91 33 ac e5 e3 ff 13 a7 de 35 57 4c 7f d2 dc b6 f8 45 f3 fc b1 73 98 54 8c 3a 0c c0 00 0c c0 00 0c c0 c0 58 33 a0 85 bb bb 96 1d 28 3e 95 f5 da 5b ef 4c a8 d5 d6 91 32 ac a5 59 3d f2 da db e6 c7 b7 ef 30 8b b6 bf 3a d6 60 8e d3 cc 91 ba b2 52 02 03 30 00 03 30 00 03 69 0c ac 7f e6 a4 7d 29 6b 9b 79 ee c8 9b 46 4f a3 27 c2 0f 10 8c 8c 61 95 d8 12 7d e7 be 57 cd f7 a7 6c 33 3b 5f 3c 63 ec 1e 63 0e 34 80 01 18 80 01 18 80 01 18 80 81 1a 03 7b 8f 7e 60 2e ba 6d 9b 59 b1 f3 b0 f9 f0 c3 0f 47 fe fb ad d9 1b d6 72 bf aa c4 9e b7 e9 80 b9 d8 be 11 b7 df 6e 01 c0 ac 62 d6 61 00 06 60 00 06 60 00 06 60 c0 cf c0 01 bb 45 e0 ca 19 4f 99 3b ec be d6 0f 3e f8 60 a4 4d 6b d6 86 b5 34 ab 12 79 ca c2 67 ec 37 55 9f 35 2f bd 69 03 c3 81 06 30 00 03 30 00 03 30 00 03 30 90 c4 c0 dd 2b 0e 99 eb 1e de 33 d2 a6 35 6b c3 aa 4f 56 c9 ac ce 5c fb bc b9 d1 7e 28 17 a3 8a 59 87 01 18 80 01 18 80 01 18 80 81 e6 0c dc bb ea 90 f9 e3 a3 7b 47 d6 b4 66 6b 58 b5 ba aa 6d 00 db 9e 79 d9 5c 7c d7 4e 63 f7 0d 73 a0 01 0c c0 00 0c c0 00 0c c0 00 0c b4 64 e0 ea 59 cf 98 47 37 bd f8 89 69 1d a5 4f 5e 65 69 58 cb 17 ac 0e be f2 66 f1 82 d5 [... cut ...] 1 14 40 01 14 40 01 14 40 01 14 40 81 ac 15 c0 b0 66 1d 1e 0a 87 02 28 80 02 28 80 02 28 80 02 28 80 61 85 01 14 40 01 14 40 01 14 40 01 14 40 81 ac 15 c0 b0 66 1d 1e 0a 87 02 28 80 02 28 80 02 28 80 02 28 80 61 85 01 14 40 01 14 40 01 14 40 01 14 40 81 ac 15 c0 b0 66 1d 1e 0a 87 02 28 80 02 28 80 02 28 80 02 28 f0 ff 01 47 0d 78 79 a5 01 f3 9b 00 00 00 00 49 45 4e 44 ae 42 60 82 FLDA
447 2024/02/01 19:35:43.993943 21644.1793 20 ECU1 FLTR FLTR 1561521 log info verbose 3 FLFI 2202457823 FLFI

```

alexmucde, 2024-02-02:
@minminlittleshrimp I think this is another issue. You have to take care, that you have the right configuration loaded in the plugin with the right AppId and CtxId.

closed 2024-02-02 by alexmucde