COVESA/dlt-viewer #425: DLTViewerPlugin message window modifies message (removes 1 space if there are 2).

vpatelv, 2024-02-15:
The message window of the DLTViewerPlugin appears to automatically format for extra spaces. This makes searches unpredictable when copying payload data from the message window.

For example.
Actual payload:
Version:  123
DLTViewerPlugin display:
Version: 123

alexmucde, 2024-02-23:
@vpatelv The following line is used to display the DLT message in the plugin:

form->setTextBrowserUncoded(msg.toStringHeader()+"<br><br>"+stringToHtml(msg.toStringPayload()));

It is displayed in a HTML widget. In HTML extra spaces are removed, that is the reason.

https://stackoverflow.com/questions/17784595/do-browsers-remove-whitespace-in-between-text-of-any-html-tag

Instead of copying the message from the plugin, please use the Ctrl+C function or the "Copy Selection to Clipboard" function in the context menu when you right click ath the selected line.

vpatelv, 2024-03-04:
Thanks for explaining the behavior.
Ideally I would have preferred to copy only the payload, but I will deal with copying the entire line. Yes we can close it, thank you.

closed 2024-03-04 by vpatelv