COVESA/dlt-viewer #426: Convert TCP Byte Stream into ASCII using "File -> Import DLT Stream"

BennoSchebitz, 2024-02-16:
Hello,
I need some support for a new functionality.
I collect TCP Byte Stream that includes DLT trace data of an ECU.

I want to convert this TCP Byte Stream into an ASCII file by using DLT-Viewer functionality to get for example header timestamp information readable. I am using Windows.

I try to use based on "dlt_viewer_user_manual.pdf" the command line functionality to convert files:
      4.8.3.4 Convert to ASCII file
      dlt_viewer -c ./example.dlt ./example.txt

Unfortunately this is limited for *.dlt files only (there it works as expected). 

What would be really helpful to get a functionality that uses "File -> Import DLT Stream" functionality and rewrite results in an *.txt file. And this should be executable also from windows CMD line. 

With this windows user could save a lot of time to writing own DLT-Trace-Parser for there logging frameworks. 

Hope anybody can support here.

BR,
Benno

alexmucde, 2024-02-23:
@BennoSchebitz You are right, the import functions like import stream are not usable from commandline. This functionality must be added.

mrclauss, 2024-05-29:
Here's a patch to allow that.

closed 2024-06-04 by alexmucde