COVESA/dlt-viewer #440: Build and Run DLT-Viewer in debug mode. Tips?

MaxMagazin, 2024-02-29:
Hi dlt-viewer Team,

I am experience a crash of the app in my macOs environment, when I try to make a new ECU/TCP connection. The app starts normally though. I assume it NOT a problem of dlt-viewer, rather of my local environment. Cause I did have it working properly some time ago, but most probably I ruined something in my environment while installing and setting up other software packages.

[image]

`Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [2691]`

How do I build and run the dlt-viewer in Debug mode to see more debug outputs, where exactly the app breaks? 
So far I found a cmake flag (CMAKE_BUILD_TYPE) to build the Debug type. But can not find the cli option to start the app with debug outputs. 
For building in debug mode I use: CMAKE_BUILD_TYPE=Debug Qt5_DIR="/opt/homebrew/opt/qt5" cmake "Xcode" ../ && make

The detailed macOS-Report is also attached. 
[2024_02_29_dlt-viewer-crash.txt](https://github.com/COVESA/dlt-viewer/files/14449260/2024_02_29_dlt-viewer-crash.txt)

Any help will be appreciated.

bunty95, 2024-03-01:
are you running the ".app" or the "unix makefile"?
Which version of DLT?

MaxMagazin, 2024-03-03:
Hi @bunty95, 
I used the makefile to build the latest available release sources (v2.25.0), available [here](https://github.com/COVESA/dlt-viewer/archive/refs/tags/v2.25.0.tar.gz)

bunty95, 2024-03-27:
@MaxMagazin Use below commands, and replace the cmake portion of build.sh in the Darwin fodler.
Let me know if you have issues.

cmake -G Ninja \
  -DCMAKE_INSTALL_PREFIX=${INSTALL_DIR} \
  -DCMAKE_PREFIX_PATH=/opt/qt512/lib/cmake \
  -DCMAKE_BUILD_TYPE=Debug \
  -DDLT_USE_QT_RPATH=ON \
  -DDLT_PARSER=OFF \
  -DDLT_APP_DIR_NAME=${APP_DIR_NAME} \
  -DDLT_LIBRARY_INSTALLATION_PATH="${APP_DIR_NAME}/Contents/Frameworks" \
  -DDLT_EXECUTABLE_INSTALLATION_PATH="${APP_DIR_NAME}/Contents/MacOS" \
  -DDLT_RESOURCE_INSTALLATION_PATH="${APP_DIR_NAME}/Contents/Resources" \
  -DDLT_PLUGIN_INSTALLATION_PATH="${APP_DIR_NAME}/Contents/MacOS/plugins" \
  "${SRC_DIR}"
cmake --build "${BUILD_DIR}"

bunty95, 2024-04-22:
@MaxMagazin any update?

alexmucde, 2024-09-09:
@MaxMagazin Still an issue? If not please close the ticket.

MaxMagazin, 2024-09-25:
Sorry for the late response. 
And for those, who accidentally stumble on this issue: as mentioned in the first post, it was rather a problem with my local environment and I fixed it somehow by reinstalling dependencies.
Closing this.

closed 2024-09-25 by MaxMagazin