COVESA/dlt-viewer #488: Notarization of MacOS artifacts

hannesa2, 2024-06-13:
Currently, we can build for MacOS, but we can't start because of missing notarization.
The idea is to do it like https://github.com/gitx/gitx/pull/334/files and for this we need a sponsor for an Apple Developer ID, or someone who fill these variables with the necessary secrets 

[image]

bunty95, 2024-06-13:
@hannesa2 I have a apple dev account and can help provide most of this info

hannesa2, 2024-06-13:
@alexmucde as long as we don't have access to this settings for variables, we can't continue here.

alexmucde, 2024-06-14:
@hannesa2 You mean access to "Actions secrets and variables"?

bunty95, 2024-07-10:
@hannesa2 can you see if this application is signed and works for you?
https://github.com/bunty95/dlt-viewer/releases/tag/2.26.0.1

hannesa2, 2024-07-10:
It's not signed jet. We wait for @alexmucde

hannesa2, 2024-07-10:
> @hannesa2 You mean access to "Actions secrets and variables"?

As discussed in a 1:1 chat: yes

bunty95, 2024-07-13:
> It's not signed jet. We wait for @alexmucde

Understood, I have signed this app, but need someone to double check on arm64. Could you check and report back for me?
https://github.com/bunty95/dlt-viewer/releases/tag/2.26.0.1 <- DLT-Mac_arm64.zip

hannesa2, 2024-08-24:
> Is it possible to let us add the secrets? I have notarization working on my fork of this repo.

@alexmucde It's you task to enable someone to do so

alexmucde, 2024-09-09:
@bunty95 @hannesa2 Please let me know what exactly what to do. We can also do a Teams session together.

hannesa2, 2024-09-09:
> @bunty95 @hannesa2 Please let me know what exactly what to do. We can also do a Teams session together.

As outcome, you need to give @bunty95 (temporary) admin access, to allow him to add credentials as a variable

alexmucde, 2024-09-10:
@bunty95  Done, please check if it works.

bunty95, 2024-09-10:
@alexmucde doesn't look like i am able to add secrets.
Looking at the GitHub docs I will need Admin access. Not sure how to do that.
https://docs.github.com/en/actions/security-for-github-actions/security-guides/using-secrets-in-github-actions

alexmucde, 2024-09-11:
Done

bunty95, 2024-09-12:
Secrets added, tried to update the PR workflow to pull in the secrets but gives me an error.
Using the same commands on my personal fork and it works.
@hannesa2 any idea what might be the issue?

hannesa2, 2024-09-12:
I see no pull request here https://github.com/bunty95/dlt-viewer/pulls

PR #532 (2024-09-13): Notarization (copy)
close #488
closed 2024-09-13 by hannesa2