COVESA/dlt-viewer #492: SOMEIP plugin was not supported in Linux dlt-viewer v2.25.0 version

wojoin, 2024-06-21:
SOMEIP plugin was not supported in the dlt-viewer v2.25.0 version. But SOMEIP plugin was provided in v2.23.0 and I can select the specific fibex version.
At the same time, It is not clear for how to use the command with SOMEIP plugin in Linux or Windows.
v2.25.0
[image]

v2.23.0
[image]

alexmucde, 2024-06-21:
@wojoin SOMEIP Plugin was never part of the COVESA distribution of DLT Viewer. Please contact me at BMW if you need more information.

alexmucde, 2024-06-21:
Please close the ticket if the question is answered.

wojoin, 2024-06-24:
I got more information from @alexmucde and it's not issue.

closed 2024-06-24 by wojoin

hannesa2, 2026-01-14:
Here is the source https://github.com/COVESA/vsomeip