COVESA/dlt-viewer #503: DLT Viewer is unable to handle files from windows network share

rohitz, 2024-07-15:
drag and drop any file (size does not matter) from windows network share. Open the file in DLT viewer and it will hang.

Force closing and restarting DLT viewer does not bring it to initial state: it will bootup hanging

Deleting %userprofile%\appdata\local\dlt-viewer\cache does not resolve this issue

Deleting DLT and all accosiate files - does not resolve this issue

alexmucde, 2024-07-23:
@rohitz Do you have Write access to the network share? If not this could be an issue.
Also regarding performance you should copy the files to a local drive. DLT Viewer is designed to not load the DLT File into memory to be able to handle big files.

rohitz, 2024-07-24:
@alexmucde Copying the file locally works. Also there is write access to the network share. I should have also mentioned the file size was around 1gb but also waited around 10 mins for the file to load. I'm not too concerned about the performance part.

Is there anything I can get like logs that can help find the issue?

alexmucde, 2024-09-09:
@rohitz Can you check again if the current version still has this issue? Need to find out what is blocking loading read only files.

closed 2025-07-24 by alexmucde