COVESA/dlt-viewer #514: Mac App Store Release

bunty95, 2024-08-01:
Since I have a developer account with Apple, should I release DLT Viewer through the Mac App store?
A link to the Mac app store can take care of distribution for signed apps along with the signed apps being posted on github.

Thoughts @hannesa2 and @alexmucde ?

hannesa2, 2024-08-01:
I at least love the idea. Best with automatic publishing on a new release

bunty95, 2024-08-01:
Sounds good, I will set it up and add the link in a PR.

bunty95, 2024-08-21:
DLTViewer is available on Appstore:

https://apps.apple.com/us/app/dlt-viewer/id6648796931

hannesa2, 2024-08-21:
Thank you. At the one hand it's a nice option for lazy guys 👍

At the other side, it's not an comprehensible build and we don't know the exact source of it. I at least, miss some version info, e.g

[image]

Unfortunately I'm no qt developer, and for me it's hard to add this. But I guess it's something like this https://github.com/nocnokneo/cmake-git-versioning-example

In general, I would love to let the version artifact built by our CI and upload it from there https://github.com/COVESA/dlt-viewer/issues/488 there we could use brew as source as well

hannesa2, 2024-08-21:
Any chance to make this work ?

> In general, I would love to let the version artifact built by our CI and upload it from there https://github.com/COVESA/dlt-viewer/issues/488 there we could use brew as source as well

closed 2024-08-21 by hannesa2

bunty95, 2024-08-21:
Its a good idea, let me run some tests and see if i can get something like that added to the appstore release.