COVESA/dlt-viewer #517: No commnandline output when starting dlt-viewer without option from commandline in Windows

alexmucde, 2024-08-08:


vifactor, 2024-09-01:
I see this output if I start the viewer (2) on Windows from cmd:
[image]

Of course due to https://github.com/COVESA/dlt-viewer/issues/508 if one wants to run in console mode, it won`t currently work unless offscreen.dll plugin is copied over from Qt installation. Once done:
[image]

vifactor, 2024-09-01:
There is nothing printed after this line: https://github.com/COVESA/dlt-viewer/blob/cd2d1ffb963c8dcfac42db3a4baf96fa05518de5/qdlt/qdltoptmanager.cpp#L292

If I remove it, I can see full output, but of course then if executable is launched not from console but via double click, the console appears.

In any case, shouldn't the log output be redirected in a log file...

alexmucde, 2024-09-09:
You are right. Command line mode is only activated in specfic use cases like, like -c conversion.
Should we introduce a new command line option to activate command line mode?

vifactor, 2024-09-09:
> Should we introduce a new command line option to activate command line mode?

do you mean that if dlt-viewer is launched as follows from a cmd like `.\dlt_viewer -c`, then we do not call `FreeConsole();` and hence `qDebug` prints messages into console? It is probably ok solution for now to give Windows users possibility to see log output, e.g. to troubleshoot an issue.

In the mean time, as I said above, it would probably be good to think about implementing file logging, since such persistent logs can be helpful for troubleshooting and bug analysis.

I'll make a PR with `-c` option soon to try

vifactor, 2024-09-10:
-c option is occupied, but I'll try another solution around this: https://stackoverflow.com/questions/9009333/how-to-check-if-the-program-is-run-from-a-console

closed 2024-09-12 by alexmucde