COVESA/dlt-viewer #523: Control message request over UDP(unicast/multicast) not working

tejaswinijoshi89, 2024-08-30:
I am connecting dltviewer (tried with v2.26.0, v2.27.0, v2.18.0) with dltdaemon (https://github.com/COVESA/dlt-daemon/tree/master) over UDP. I am able to receive log messages from dltdaemon over both UDP unicast and multicast interface. But when I try to send control message from dltviewer (GetLogInfo,GetDefaultLoginfo,etc) i get error as "ECU Not Connected ". I even checked on wireshark. UDP signal transmission traces from dltviewer to dltdaemon over UDP are not present.
Does dltviewer supports control message request over UDP. If so then how to enable it.

vifactor, 2024-09-03:
I tried to have a look to this issue and cannot even configure dlt-daemon to be able to receive dlt logs via UDP in dlt-viewer. I run daemon with `src/daemon/dlt-daemon -c custom_dlt.conf`, where the conf file contains a single entry: `UDPConnectionSetup = 1`. On dlt-viewer side I connect with:
[image]

The connection to ECU does not happen (connection icon is yellow, not green). If I switch to TCP, I can connect and see some technical logs. what do I miss for UDP?

alexmucde, 2024-09-09:
Sending control messages is only supported for TCP connections. Sending UDP messages to ECUs is not allowed in our cars. I do not know how other OEMs are handling this.

alexmucde, 2024-09-09:
DLT Daemon also does not support this, because of the same reason.

Samuel-John, 2024-11-06:
Hi @tejaswinijoshi89 / @vifactor ,
Were you able to connect to ECU via UDP?

@alexmucde , I am facing the same issue. The UDP connection is not happening to ECU. It remains in yellow. Due to this, the DLT messages from ECU are not seen on the dlt-viewer.

Do you have any workaround for this?

alexmucde, 2024-11-06:
Hello together,
i would like to know exactly in which version and on which platform UDP reception is not working anymore.
I have got another report that there is an issue with Windows 11.
- Qt version used
- OS used (Linux, Windows 10/11, MacOs)
- DLT Viewer version used
It would be also important with which version/commit this issue happened the first time to identify a possible commit, which causes the regressions.

vifactor, 2024-11-06:
I tried multiple version of the dlt-viewer. In particular 2.21, 2.26 and the one built from master. Never could connect to dlt-daemon (built from scratch and running directly on my machine) even if I configure it to use UDP as I described above. I use ubuntu. Qt 5.15.x and 6.4.x

Firewall can be indeed the reason, but I am not sure how to check it.

alexmucde, 2025-07-24:
Issue is answered and solved.

closed 2025-07-24 by alexmucde