COVESA/dlt-viewer #524: Set log level

aserag1, 2024-09-03:
Hello, 
I am trying to change set log level, but it always dimmed if u can help me how can i use it .
[image]

vifactor, 2024-09-04:
After looking into the code for ~1h, I figured out how and managed to make that menu item active. For that you need:
1. to be connected to an ECU
2. in the "Config"-tab with your ECU, unfold the tree-view with appIds and contextIds, then select a tree leaf

Open the "DLT" app menu item and the "Set Log Level" entry should be active. 

P.S: I do not understand how this feature suppose to work, though...

aserag1, 2024-09-05:
can u please provide photoes for in the "Config"-tab with your ECU, unfold the tree-view with appIds and contextIds, then select a tree leaf

vifactor, 2024-09-05:
Here is a screenshot:
[image]

alexmucde, 2024-09-09:
The feature should be only working, when there is an active connection to an ECU by TCP. UDP will not work.

aserag1, 2024-09-09:
there is always an active connection by TCP and not active with me either

alexmucde, 2024-09-10:
As @vifactor already mentioned, a context must be selected in the Project/Config View.
[image]

alexmucde, 2025-07-24:
Close ticket as it was answered.

closed 2025-07-24 by alexmucde