COVESA/dlt-viewer #547: 2.26.1 release file for macOS is empty

oxoxoxox, 2024-09-29:


hannesa2, 2024-09-29:
Unfortunately yes, but the corresponding master build comes with the binary

oxoxoxox, 2024-09-29:
> Unfortunately yes, but the corresponding master build comes with the binary

OK, I already download the master build binary from Actions

hannesa2, 2024-09-29:
But now https://github.com/COVESA/dlt-viewer/actions/runs/11093293490 I see the notarization and codesign is missing  in release.
@bunty95 just to inform you

hannesa2, 2024-09-29:
Now the release job includes the artifacts https://github.com/COVESA/dlt-viewer/actions/runs/11093436417/job/30819891681, but the release job itself miss it

closed 2024-09-29 by hannesa2

hannesa2, 2024-09-29:
Finally the MacOS artifacts are in release https://github.com/COVESA/dlt-viewer/releases/tag/2.26.2 now
But it includes a lot of noise https://github.com/COVESA/dlt-viewer/issues/534

PR #548 (2024-09-29): Fix MacOS release artifacts
It's hard to test a release without making a release, so I will create a new release

close #547
PR #550 (2024-09-29): Include mac artifacs release
close  #547