COVESA/dlt-viewer #557: Only one build system instead of two

alexmucde, 2024-10-11:
Currently the DLT Viewer supports two build systems, cmake and qmake.
To reduce the maintenance effort and the possiblility of wrong configurations this should be reduced to one, preferable cmake.
It must be taken care, that all development use cases must be supported:
- Build Scripts for Windows
- Build Scripts for Linux Debian packages (uses qmake)
- Development, Debug and Test with QtCreator (uses qmake)
- Github Action CI

hannesa2, 2024-10-11:
From the CI point of view:
* Windows builds already with [cmake](https://github.com/COVESA/dlt-viewer/blob/f8483534aa98bb9b9c6a2e9c9feac6868e4936c4/.github/workflows/BuildPR.yml#L160)
* MacOS builds already with [cmake](https://github.com/COVESA/dlt-viewer/blob/f8483534aa98bb9b9c6a2e9c9feac6868e4936c4/.github/workflows/BuildPR.yml#L41)
* Linux builds with [qmake](https://github.com/COVESA/dlt-viewer/blob/f8483534aa98bb9b9c6a2e9c9feac6868e4936c4/scripts/linux/build.sh#L20-L22) and [cmake](https://github.com/COVESA/dlt-viewer/blob/f8483534aa98bb9b9c6a2e9c9feac6868e4936c4/scripts/linux/build.sh#L30-L43) I've no clue if we simply can remove the qmake stuff
* I've no clue about _"Development, Debug and Test with QtCreator"_ at least the CI don't use it (as far as I know). That means it's not tested (and supported ?)

hannesa2, 2024-10-11:
With https://github.com/COVESA/dlt-viewer/pull/562 the CI is qmake free

hannesa2, 2024-10-11:
> With #562 the CI is qmake free

After merge it, the CI is now qmkae-free

alexmucde, 2025-07-24:
Issue is under work.