COVESA/dlt-viewer #576: UI is unresponsive fo seconds at the end of dlt load

vifactor, 2024-11-12:
Long running MainWindow::reloadLogFileFinishFilter() is executed in main thread and hangs UI

vifactor, 2024-11-23:
This https://github.com/COVESA/dlt-viewer/blob/3f51d3f71145c5a5407a969f9cfebe620275abca/src/mainwindow.cpp#L2068 helps to avoid UI freeze, but unfortunately it is not thread safe as far as I can see and currently emits a lot of warnings.

The real fix as I can see it will include significant refactoring of the `MainWindow::contextLoadingFile` function.

vifactor, 2024-11-27:
Refactored code a bit here.

Apparently, parsing messages takes much less than populating the tree, i.e. constructing and updating UI elements
getLogInfoMessages took  196 ms
populateEcusTree took  4309 ms

vifactor, 2024-11-29:
@alexmucde currently when I import 2 dlt files sequentially, in the logs table only the logs from the second file become seen. I would expect that in such case the ECUs tree from the "Config"-tab should correspond to the latest imported file. This doesn't however happen, the ECUs tree gets appended as far as I can see, so it contains ECUs data from both files. Is this bug or feature?

alexmucde, 2024-12-11:
@vifactor Issue solved and integrated, can be closed?

vifactor, 2024-12-11:
@alexmucde issue is not resolved, the PR you merged is a little improvement. The real fix is more complicated, I have a draft for it, but needs cleanup. Please keep this issue open

If possible, could you give me the answer for the question above: https://github.com/COVESA/dlt-viewer/issues/576#issuecomment-2508651277 ?

vifactor, 2025-02-01:
@alexmucde the PR is in place, it fixes the issue. I am not sure, though I tested it well for bad inputs, since I do not have a collection of test cases with various inputs. Maybe we can merge it, people start using it and when/if there is a regression found we fix it? I do not see any other way to validate that not-so trivial PR

alexmucde, 2025-02-17:
Following pull request can solve the issue, please try:
https://github.com/COVESA/dlt-viewer/pull/633