COVESA/dlt-viewer #599: Sort dlt files in the proper sequence

oxoxoxox, 2024-12-06:
I have some dlt files, which is named as xxx.1.dlt, xxx.9.dlt, xxx.10.dlt, xxx.99.dlt, xxx.100.dlt ...

Can author modify the sourcecode to sort these files in the proper sequence of increasing numbers?
And when I select the folder to open all dlt files, these files can also be opened in this sequence.

PS: In macOS, Finder can sort these files properly.

[image]

alexmucde, 2024-12-06:
Good idea. I could add easily add a sorting of the filenames when opening, which should be normally sone by Windows Explorer but we still have issues with sorting like with DN_KRNL.12.dlt. Better naming would be DN_KRNL.012.dlt.
But this what you think should be implemented, that also DN_KRNL.12.dlt is handled correctly in the right order?

oxoxoxox, 2024-12-07:
> Good idea. I could add easily add a sorting of the filenames when opening, which should be normally sone by Windows Explorer but we still have issues with sorting like with DN_KRNL.12.dlt. Better [...]

Yes, I think DN_KRNL.12.dlt should treat as DN_KRNL.012.dlt while sorting. Just like 001 to 999.

oxoxoxox, 2024-12-07:
And yes, write the filename as DN_KRNL.012.dlt would be a better solution, but this is not what I can determined.🤣

alexmucde, 2025-07-24:
I could think about sorting the files by filenames before loading, but the issue you are describing is very special.
I would propose to provide the files already to DLT Viewer in the right order.

closed 2025-07-24 by alexmucde