COVESA/dlt-viewer #612: Cmake issue and How to create DLT installer package

oblesh6, 2025-01-08:
Hello,

I have followed the instructions mentioned in the file \dlt-viewer-2.25.0\scripts\windows\build.md 

I have successfully able to build the source code

But when I execute "cmake install . --prefix C:\DltViewerSDK", i'm getting an error as shown below

>cmake install . --prefix D:\DltViewerSDK
CMake Warning:
  Ignoring extra path from command line:

   "D:\DltViewerSDK"

CMake Error: Unknown argument --prefix
CMake Error: Run 'cmake --help' for all supported options.

what is the actual issue here? Also I want to create Installer package.

Please let know any solutions

Thanks :)

alexmucde, 2025-01-08:
Please look at the bat files for Windows to build the DLT Viewer.
The Installer package generation we currently do not use, perhaps they are not up to date.

vifactor, 2025-01-19:
@oblesh6 there is either a misprint in the docs or that command was written for a specific build system (not the same as you use). I am not sure if you really want to install the binaries you built yourself, but if you do, please try the following
```
cmake --install <build_dir> --prefix <your_prefix_path>
```

Alternatively, you should be able to control install prefix by passing `-DCMAKE_INSTALL_PREFIX=[..]` at configuration step

closed 2025-03-16 by vifactor