COVESA/dlt-viewer #616: Search matches if payload of message is empty

alexmucde, 2025-01-21:
Currently also DLT messages with empty payload will be shown in the search results.

vifactor, 2025-01-21:
is this something I can help with? does it show messages with empty payload when searching empty messages, i.e. query string is empty? 

would be nice to have a test dlt file if possible

alexmucde, 2025-01-21:
Yes  you can help here. I have created a Tool for another purpose, but this generates some DLT messages, also empty ones with some modifications. I have added this file. If you now search for Relais3, you get also the empty messages in the search results.

[20250121_140442_DLTRelaisPluginTest_Logs.zip](https://github.com/user-attachments/files/18491280/20250121_140442_DLTRelaisPluginTest_Logs.zip)

vifactor, 2025-01-22:
So far cannot reproduce:

[image]

Tried on Linux v.2.27.0 and current master head. Will try Windows later. Maybe you have some non-default settings?

alexmucde, 2025-01-23:
Hmm my output looks like this, including my Search settings in the Screenshot:

[image]

alexmucde, 2025-01-23:
When Regular Expressions is active, the empty lines are gone.

vifactor, 2025-01-23:
thanks, indeed with regular expressions off could reproduce. Will try to find time to understand why this happens

vifactor, 2025-01-24:
@alexmucde PR with fix is up