COVESA/dlt-viewer #620: DLT Viewer user interface currently very slow on Ubuntu 24.04 Noble Qt5

alexmucde, 2025-01-29:
The DLT Viewer user interface currently reacts very slowly when scrolling or trying to select and message in the table windows. This happens to me since some weeks, possible some updates of the Qt5 libraries.

Does any other users have similar observations?

alexmucde, 2025-01-31:
Looks like this is an issue only with VNC and TurboVNC. If i run it with a local monitor i see no obivious issue.

beidouchangxi, 2025-03-21:
I experienced the similar issue on Windows 10 + DLT 2.28.0 which compiled with MSVC 2019 + QT6.

The UI experience is acceptable (but not very well) on my HP ZBook 15.6 G8, and significantly slow on another HP E830 G8, espacially works with my own plugin.

When switch back to v2.25.0, it become fluent again.