COVESA/dlt-viewer #640: Copy selected table rows as a screenshot

vifactor, 2025-03-14:
Add new table context menu item in addition to (copy to clipboard, copy into JIRA) "Copy as image".

This is often used to insert in JIRA comments as an evidence of a problem. The "copy to JIRA" context menu item is limited because the inserted table does not preserve the coloring of rows and in general the JIRA table formatting is not very convenient to look at.

beidouchangxi, 2025-03-21:
Maybe it's better to copy HTML format rather than screenshot?

vifactor, 2025-03-21:
> Maybe it's better to copy HTML format rather than screenshot?

can be yet another option, yes. I am interested in having screenshot because this is what I usually use to share my log findings with colleagues. 

It is easy to use 3dparty for making screenshot, but why doing so if it can be done withing the app?

closed 2025-11-02 by vifactor